Dukweeno/LocaleKeyboard.js #2: The es_ES.lang is totaly invalid and incomplete.

Thecakeisgit, 2016-10-26:
We need somebody with a spanish keyboard to fill the file with correct ascii map.

d4n5h, 2016-10-26:
@Thecakeisgit I'm impressed by you. Good job mate.
You're becoming a great maintainer.

d4n5h, 2016-10-26:
@Thecakeisgit I don't have a French or a Spanish keyboard. just a regular QWERTY.
So i can't help you with that one.
Sorry.

d4n5h, 2016-10-26:
@Thecakeisgit By the way, i've referenced this project on my repository if you don't mind.

d4n5h, 2016-10-26:
@Thecakeisgit http://www.ascii-codes.com/cp852.html Check out this site. it's got Ascii tables for almost all languages.

d4n5h, 2016-10-26:
@Thecakeisgit https://github.com/NicoHood/HID/issues/22 Check this one out.
It's a good place to ask for help with testing and adding additional layouts.

Pedrinbeep, 2016-10-26:
@Thecakeisgit I have spanish keyboard. I am very noob but tell what to do.

Thecakeisgit, 2016-10-26:
Thanks @d4n5h :D i decided to make good maintainement for this and Dckuino.js ;D
@Pedrinbeep I'll tell you tomorrow, i'm really tired, i go to sleep ;D I think i can do it also, i'll just need you to test ^^

Pedrinbeep, 2016-10-26:
@Thecakeisgit Great! Thank dude!! ;)

Thecakeisgit, 2016-11-02:
@Pedrinbeep we finally fixed Dckuino.js, so now we just need to update .lang files ;D

Pedrinbeep, 2016-11-03:
@Thecakeisgit Nice! Tell me what to do for make any test you want on a spanish keyboard, and how to do it please. ;)

Pedrinbeep, 2016-11-21:
Hi @Thecakeisgit 
I finally try your library with my spanish keyboard and your keyboard.h spanish layout and something ins wrong. The arduino sould write this:

`abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ0123456789\\ !#$%&'`*~{}[]()%,;:!?-_./+=<>\"\\`

And it write this:

`abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ0123456789ç !·$%/´º(ª^*`+)=%,ñÑ!_'?.-¿¡;:¨ç`

That is all I can help. Thank you.

PD: Do you know if your Dckuino.js can also work with a Teensy? Just for fun.

Thecakeisgit, 2016-11-21:
Hi @Pedrinbeep, it's been a while :D, Dckuino.js work with teensy, but the layout is messed up and you can't use the modfied Keyboard.h

Thecakeisgit, 2016-11-21:
@Pedrinbeep For the layout, you made the keys, it's cool, i'll see if i can do something ;D (Do u have a picture of ur keyboard ? xD)

Pedrinbeep, 2016-11-21:
Hi @Thecakeisgit ! Perfect, I will also try it on Teensy with their own keyboard. 

About my layout keyboard, all yours! Hahaha. Thank you so much.

[image]

Borja-Garduno, 2017-01-15:
@Thecakeisgit it´s been a while since the last comment, however I just find this git topic. I´m Spanish so my keyboard layout was also mess up. I have I think achieve almost every key (by mixing some of the langs and trying it out), but I´m not capable of printing neither of: | @ # ^.

My Keyboard.cpp looks like:
`/*
  Keyboard.cpp

  Copyright (c) 2015, Arduino LLC
  Original code (pre-library): Copyright (c) 2011, Peter Barrett

  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General Public
  License along with this library; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
*/

#include "Keyboard.h"

#if defined(_USING_HID)

//================================================================================
//================================================================================
//  Keyboard

static const uint8_t _hidReportDescriptor[] PROGMEM = {

  //  Keyboard
    0x05, 0x01,                    // USAGE_PAGE (Generic Desktop)  // 47
    0x09, 0x06,                    // USAGE (Keyboard)
    0xa1, 0x01,                    // COLLECTION (Application)
    0x85, 0x02,                    //   REPORT_ID (2)
    0x05, 0x07,                    //   USAGE_PAGE (Keyboard)

  0x19, 0xe0,                    //   USAGE_MINIMUM (Keyboard LeftControl)
    0x29, 0xe7,                    //   USAGE_MAXIMUM (Keyboard Right GUI)
    0x15, 0x00,                    //   LOGICAL_MINIMUM (0)
    0x25, 0x01,                    //   LOGICAL_MAXIMUM (1)
    0x75, 0x01,                    //   REPORT_SIZE (1)

  0x95, 0x08,                    //   REPORT_COUNT (8)
    0x81, 0x02,                    //   INPUT (Data,Var,Abs)
    0x95, 0x01,                    //   REPORT_COUNT (1)
    0x75, 0x08,                    //   REPORT_SIZE (8)
    0x81, 0x03,                    //   INPUT (Cnst,Var,Abs)

  0x95, 0x06,                    //   REPORT_COUNT (6)
    0x75, 0x08,                    //   REPORT_SIZE (8)
    0x15, 0x00,                    //   LOGICAL_MINIMUM (0)
    0x25, 0x65,                    //   LOGICAL_MAXIMUM (101)
    0x05, 0x07,                    //   USAGE_PAGE (Keyboard)

  0x19, 0x00,                    //   USAGE_MINIMUM (Reserved (no event indicated))
    0x29, 0x65,                    //   USAGE_MAXIMUM (Keyboard Application)
    0x81, 0x00,                    //   INPUT (Data,Ary,Abs)
    0xc0,                          // END_COLLECTION
};

Keyboard_::Keyboard_(void)
{
  static HIDSubDescriptor node(_hidReportDescriptor, sizeof(_hidReportDescriptor));
  HID().AppendDescriptor(&node);
}
[... 267 lines cut ...]
void Keyboard_::releaseAll(void)
{
  _keyReport.keys[0] = 0;
  _keyReport.keys[1] = 0;
  _keyReport.keys[2] = 0;
  _keyReport.keys[3] = 0;
  _keyReport.keys[4] = 0;
  _keyReport.keys[5] = 0;
  _keyReport.modifiers = 0;
  sendReport(&_keyReport);
}

size_t Keyboard_::write(uint8_t c)
{
  uint8_t p = press(c);  // Keydown
  release(c);            // Keyup
  return p;              // just return the result of press() since release() almost always returns 1
}

Keyboard_ Keyboard;

#endif
`

Any ideas?

Thecakeisgit, 2017-01-15:
Hi, :D I'm happy too see that you were working on this project :) 
I'm pretty sure that you problem is this that you use <code>|SHIFT</code> on these 4 not working keys ^^
You can't use SHIFT **and** ALTGR at the same time ;D

*NOTE: Keep in mind that ATmega32U4 is purely emulating a Keyboard, so to do #, |, @ and ^, you need to ALTGR + \<key\>, **not** ALTGR + SHIFT + \<key\> :)*

Thecakeisgit, 2017-01-15:
@Borja-Garduno If you want to you can make a pull request to add your work to the project, it would be great ;D 
Help yourself with -> [Wiki : How to create a locale ?!](https://github.com/Thecakeisgit/LocaleKeyboard.js/wiki/How-to-create-a-locale-%3F!)

Borja-Garduno, 2017-01-15:
@Thecakeisgit So, which #define should I need for ALTGR? I can see in Keyboard.h the following ones:
#define KEY_LEFT_CTRL   0x80
#define KEY_LEFT_SHIFT    0x81
#define KEY_LEFT_ALT    0x82
#define KEY_LEFT_GUI    0x83
#define KEY_RIGHT_CTRL    0x84
#define KEY_RIGHT_SHIFT   0x85
#define KEY_RIGHT_ALT   0x86
#define KEY_RIGHT_GUI   0x87

However in Keyboard.cpp 0x80 is already used by the Shift key, which is KEY_LEFT_CTRL from above. Which 0xYY should it be?

Borja-Garduno, 2017-01-15:
Ok, I have realised that I did not need to put anything on the code (for example 0x1e|ALT_GR, // | ). 
#, @ and | now works. I guess it works due to the _altGrMap[n] from below, though I do not completely understand it. I mean, both 0x1e,   // | and 0x1e,  // 1 have the same code. How does it differentiate them?

Also, I´m still stuck with ^and `, which in the Spanish Keyboard is in the same key, as you can see in @Pedrinbeep picture. 

I try this, without success:
0x2f,     // [
0x2f,     // ^
0x2f,    // `

Following the same idea from above I thought it might work. 
^, however, need shift + key + space. Any ideas?
Sorry for bothering you :)

Thecakeisgit, 2017-01-16:
Hi again @Borja-Garduno : 
  - Yes, you don't need to add any **ALTGR** in the code, because i don't implemented AltGR this way... It's a bit technical and the **_altGrMap** is a var i totally added myself.. It can be implemented like shift because **SHIFT** works with bit wise operations, and in **uint8_t** we have only **256**(0-255) possibilities.. So when you add <code>|SHIFT</code>, it means : "Add 0x80 bit to the total." and **0x80(hex) => 128(dec) => 10000000** and when the code checks for <code>|SHIFT</code>, it search for this bold bit : **1**0000000. So if i wanted to add <code>|ALTGR</code>i needed to `#define ALTGR 0x100` (Which is the next bit, but this bit doesn't exists in 8bit.)

  - For `0x1e, // |` and `0x1e, // 1`, the trick is in the **_altGrMap**. When you type **1**, you hit the **1** key, but when you type **|**, you hold AltGr and you hit the **1** key. So your Arduino does the same :D

  - Then for **^** and **\`** when you want to type **^** you hold Shift key and you hit **\`**, so your code might be like this : 
```c
0x2f, // [ <-- For this you defined the _altGrMap
0x2f|SHIFT, // ^ <-- For this you do SHIFT + `
0x2f, // ` <-- For this you just hit `
```

Hope this helps you ;D

Borja-Garduno, 2017-01-16:
@Thecakeisgit Well, I just tried what you advised, and it isn´t working. I guess that in GB keyboard it is difference. In the Spanish layout, we need to push SHIFT + key + SPACE. It is when space is pressed that the characters appear. Also, the _altGrMap (I guess) won´t work as the correct characters doesn´t appear.

Any ideas how to hit space in the arduino? As soon I achieve this two keys, I will make a pull request (if I guess how to do it :P ).
Thanks for your help so far.

Thecakeisgit, 2017-01-18:
Yes, we need to make an **_spaceMap** I guess ;D @Borja-Garduno  
(For the pull request refer to the wiki :D)

Borja-Garduno, 2017-01-18:
@Thecakeisgit Would we need to create a new function for this, or it would be included in initAltGr?
  _spaceMap[255] = true; // ESPACIO
Like this?
I will try to make now the pull request, with the updates so far.

Thecakeisgit, 2017-01-18:
Yes (we need to create a function and initialize the var) and to work we need to modify the code who is called when typing something ^^ @Borja-Garduno

closed 2017-01-18 by Thecakeisgit