Dukweeno/LocaleKeyboard.js #3: No web interface for the plugin, need somebody

Thecakeisgit, 2016-10-28:
There is an index, but we need somebody to make a real page ;D

d4n5h, 2016-10-28:
@Thecakeisgit Sure but why do we need a JS/HTML interface?
I thought we decided to approach it with pure CPP with the keyboard.init(LANGCODE) function.

d4n5h, 2016-10-28:
@Thecakeisgit I think we should just put all the languages in the same directory and let people decide which language to use from the Arduino code. so they will be able to switch from language to another language without downloading different libraries.

Thecakeisgit, 2016-10-28:
@d4n5h Yes i'm sure, look the lib, it's a js lib, i thinked about simple CPP, but we cant ifstream inside an arduino *-* so... It's impossible, and i tested, the _asciimap MUST BE CONST, otherwise it will fail :/
And bundleing all _asciimap in the lib is not good..

d4n5h, 2016-10-28:
@Thecakeisgit What about a pointer to constant?

d4n5h, 2016-10-28:
@Thecakeisgit or const std::var<int> function(){....};

d4n5h, 2016-10-28:
@Thecakeisgit You can just leave the _asciimap array empty and push to it afterwards with pointers.

Thecakeisgit, 2016-10-28:
@d4n5h Hmm but open file is not possible..
I made a generator, its proper and i can add it to Dckuino.js, i generate and pull all in zip file ;D

d4n5h, 2016-10-28:
@Thecakeisgit Let's try to figure it out. i'm sure there is a more elegant approach.
I think we should just use .cpp for the .lang files and then the compiler will load them.

d4n5h, 2016-10-28:
@Thecakeisgit This library can be really useful for a lot of projects rather than just Duckuino and the more flexible we can make it - the better.

Thecakeisgit, 2016-10-28:
Yep, but try doing what you say in another branch ;D I'll be happy to see ^^

d4n5h, 2016-10-28:
@Thecakeisgit The idea of `Keyboard.layout(LAYOUT_CODE);` is really good and i'm sure that there is a solution to do just that. You said you like challenge? so here is a challenge ;)
Why should i create another branch? we're trying to solve the same problem...
I'll help you if you're having trouble with coding in CPP (Not that i'm an expert, i suck at CPP lol).

Thecakeisgit, 2016-10-28:
Okay ^^ I'll think about it :3 i love headers file, i'll do it :D

closed 2016-10-28 by Thecakeisgit

Thecakeisgit, 2016-10-28:
(I rolled back ;D i'll try doing this ^^)

Thecakeisgit, 2016-10-28:
@d4n5h