Dukweeno/LocaleKeyboard.js #4: Can't modify const _asciimap !

Thecakeisgit, 2016-10-28:
@d4n5h Please help \*-\*

d4n5h, 2016-10-28:
@Thecakeisgit 
Example:
const int a = 0;
*(int *)&a = 39;
printf("%d", a);

Thecakeisgit, 2016-10-28:
Doesn't works try it ;D

Thecakeisgit, 2016-10-28:
Basically const is like #define to the compiler.. so...

d4n5h, 2016-10-28:
@Thecakeisgit Are you sure _asciimap *MUST* be a contant? it sounds weird...

d4n5h, 2016-10-28:
@Thecakeisgit I'll try to find a solution, i think we can solve it.

Thecakeisgit, 2016-10-28:
Yep try to remove the const declaration on _asciimap.. And try with ur hardware, just super weird..
azerty -> ebµ

Thecakeisgit, 2016-10-28:
@d4n5h

d4n5h, 2016-10-28:
@Thecakeisgit Yeah i'm working on it, i'm trying a few things. I'll update you if i'll get it to work.

Thecakeisgit, 2016-10-28:
Okay, i backuped my js compiler, so i'll push it again if you don't find :S

d4n5h, 2016-10-29:
@Thecakeisgit I think there is a workaround:
The idea is to define all the layouts in the Keyboard.cpp file and then use press/release function to switch between the layouts... I think i should work.
Similar to this: http://forum.arduino.cc/index.php?topic=245140.msg1754197#msg1754197

d4n5h, 2016-10-29:
@Thecakeisgit 
Another "dirty" option (if it's a Windows machine) is to use the Powershell to switch to qwerty and then revert back to the previous layout.

// Keep a the variable for switching back to the previous layout:
$currentLangAndKeyboard = (Get-WinUserLanguageList).InputMethodTips

// Make the switch:
$langList = New-WinUserLanguageList en-US
$langList[0].InputMethodTips.Clear()
$langList[0].InputMethodTips.Add('0409:00000409') # English (United States) - US Qwerty
Set-WinUserLanguageList $langList

It's not a "clean" solution but it will work, and on macOS it's even easier.
There should be a similar solution for Linux machines.

Thecakeisgit, 2016-10-29:
-Thecake killing himself- 
@d4n5h The "dirty" option is very... Dirty (Yes thanks captain obvious), and there is not another Lib, it's just an improvement for the base compiler.. So i got some questions : 
- How do you handle this if the computer have no other locale, and the admin have removed all other ? 
- It loses time, and for a payload, losing 7-8 sec is not good..
- It's impossible on linux, because we need a root acces to generate a locale, and it take about 20sec to generate..

Thecakeisgit, 2016-10-29:
@d4n5h For the press() workaround, please try ;D

Thecakeisgit, 2016-10-29:
@d4n5h I just tested, and modifying the var name in press and release don't work, and it's because the var _asciimap needs to be in PROGMEM, and it's why it needs to be const.
Look: 
```c
extern
const uint8_t _asciimap[128] PROGMEM;

#define SHIFT 0x80
const uint8_t _asciimap[128] =
{
	0x00,             // NUL
        /* Truncated */
```
The line with extern is the problem, and if we remove it, it types random keys..

Thecakeisgit, 2016-10-30:
@d4n5h Poke !

d4n5h, 2016-10-31:
Hi @Thecakeisgit
At the moment i really don't have any free time to help you with this project.
If you want to push it with the JS generator ("as is") i'm fine with it.

closed 2016-11-02 by Thecakeisgit