Dukweeno/LocaleKeyboard.js #5: please help !

TradeCrafter, 2016-11-17:
hello, that's an great job, on DuckyDuino, my name is Leo from Rio de Janeiro, i'm looking for an way to get my keyboard right, since we here in Brazil type in an ABNT2 keyboard, there is some extra keys, like Ç... 
the index.HTML doesnt work, i got some scan codes for ABNT2 that i've found Googling around, i'm just not 100% of sure if its the right ones...
Please, help me creating pt-BR Keyboard for my Ducky !!

Thecakeisgit, 2016-11-17:
Hi @TradeCrafter, I’m happy to see that you like my work.
For the index.html, it doesn't work because we don't need it yet and i created LocaleKeyboard.js to use with Dckuino.js.
For the pt_BR.lang, feel free to make a pull request with the file.

To test the layout, you may use this code in your Arduino IDE : 
```c
#include "Keyboard.h"

void setup()
{
  Keyboard.begin();

  delay(500);

  Keyboard.print("abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ0123456789\\ !#$%&'`*~{}[]()%,;:!?-_./+=<>\"\\");

  Keyboard.end();
}
void loop() {}
```

Please understand that it's not easy to create a key map and I raged a lot to make it for fr_FR.
So to add a key map to this project, you need to create a .lang based on the en_US.lang(Simple copy/rename it) and you need to add it to to the lang/localeList file, without the .lang(Take a look in the file if you want).
And you need to test 1 by 1 codes in the <code>_asciimap</code> to make it work. I you want to do some altGr trick, just add a <code>_altGrMap[<number of char in _asciimap>] = true;</code> to the <code>void initAltGr()</code> function.

<b>Feel free to comment here, if u need help ;D</b>

*Please note that we can't add more keys in the _asciimap var, because the HID.cpp is coded to accept only the 128 first ascii chars..*

Thecakeisgit, 2016-11-17:
Make sure to download the good Keyboard lib using Dckuino.js or with js commands ;D Some things are modified to handle altGr key combos ;D.

TradeCrafter, 2016-11-17:
Thanks for the fast response !!
i know for sure this code you've put on the answer will not output the right codes, i dont intend to use altGR tricks and combos... just the plain regular ascii fot pt-BR.
I managed to interact with the Windows in order to change the input keyboard layout, but its only effective if the system got the lang pack already installed.
so the best solution is to create an lib for pt-BR keyboard, like the Teensy got... have you seen this ?

So let me see if i understand: i need to get yours "en_US.lang" and just trade the scan codes for the brazilian ones, once i did that where to put this file ? i'll do this .lang file right now, can i upload for you apreciate if its okay ?

Thecakeisgit, 2016-11-17:
@TradeCrafter Yep you need to deal with scan codes to match with your keyboard, and put your pt_BR.lang in lang dir and add <code>pt_BR</code> in the <code>localeList</code> file.
For the HID.cpp, we can do it, but making a HID.cpp generator plus a Keyboard.cpp generator.. I think it's not good..
The best way to help, it's to fork this repo and send a pull request once you done it ;D

*NOTE: LocaleKeyboard.js is just a Keyboard.cpp/.h generator..*
So you need to generate a version of Keyboard.cpp and Keyboard.h with Dckuino.js by entering : 
```
STRING abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ0123456789\ !#$%&'`*~{}[]()%,;:!?-_./+=<>"\
```
 in https://thecakeisgit.github.io/Dckuino.js/ and clicking the <b>Download Sketch</b> button.
Once you done this, select a locale(I recommend en_US), and finnaly click <b>Start Download</b>.

Now open you Arduino IDE and play with values in Keyboard.cpp to make the string written by your Arduino more identical to the string entered.

TradeCrafter, 2016-11-17:
STRING abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ0123456789\ !#$%&'`*~{}[]()%,;:!?-_./+=<>"\

"Download Sketch" button doesnt work... it does nothing, and this will only give an sketch to make the arduino print, somewhere on the screen these US chars, and thats it...  doesnt generate an Keyboard.H/CPP

i just realize that the characters for the ABNT2 Keyboard belongs to the 128-255 ascii group of chars

i think thats the end line... i'll never get and arduino working as an pt-BR Keyboard.

Thecakeisgit, 2016-11-17:
You need to click <b>Compile</b> before using <b>Download Sketch!</b> ;D

Thecakeisgit, 2016-11-17:
And not every keys are in 128-255 ASCII group.. You may try to convert for you locale ;D
*(I make Dckuino.js, follow me xD It generates a Keyboard.cpp/Keyboard.h)*

Thecakeisgit, 2016-11-17:
@TradeCrafter

TradeCrafter, 2016-11-17:
i know this man, by clicking COMPILE, it generates the sketch on the output frame, but the download button doesnt work, 
locale.js the same, how to run locale.js, if the index does nothing ?

please i dont want to be and pain in the ass, just need to understand in order to get my DuckyBR working !

Thecakeisgit, 2016-11-17:
You can't help the project, cheers ! @TradeCrafter

closed 2016-11-17 by Thecakeisgit

TradeCrafter, 2016-11-17:
i'm lost for words... good luck in life bro

Thecakeisgit, 2016-11-17:
@TradeCrafter I wanted to say sorry, my rage level was up to mlg/10 :( Bye, if u wanna help again reopen the issue... I didn't thinked before writing...