Dukweeno/LocaleKeyboard.js #6: Tutorial on how to create or to modify layouts

gh7531, 2016-11-21:
I would love to help by providing a german keyboard layout but I have no idea how to do that. 
A tutorial on how to create or to modify layouts and also on how to use any of the stuff on this rep would be nice.

Thecakeisgit, 2016-11-21:
Hi @gh7531, I like the idea to create a tutorial, I'll do this very quickly.. Do you think I need to do it on the README.md or on a Wiki or what else ? Please tell me ;D

gh7531, 2016-11-21:
I would use the wiki and then link to the wiki in the readme file. But I'm no expert regarding github :)

Thecakeisgit, 2016-11-21:
Done @gh7531 :D 
Wiki : [How to create a locale ?!](https://github.com/Thecakeisgit/LocaleKeyboard.js/wiki/How-to-create-a-locale-%3F!)

Any question ?

Thecakeisgit, 2016-11-22:
The problem is the Arduino HID.cpp code, coupled to the Arduino Core who is made for only ascii 128.. @gh7531

gh7531, 2016-11-22:
So extended ascii and therefore letters like the german umlauts ä,ü,ö are impossible?

Thecakeisgit, 2016-12-03:
I don't think so.. @gh7531 (Sry for the late ;D) 
Can i close the issue ?

closed 2016-12-04 by gh7531