Dukweeno/LocaleKeyboard.js #14: backslash isn't working in de_DE layout

Nibot1, 2016-12-11:
If I try to print an backslash "\" my arduino prints an "ß"
Does anybody has got an idear ?

gh7531, 2016-12-11:
I will look into it as soon as I get my arduino working again

gh7531, 2016-12-11:
So I got my Arduino working again and all worked fine. I provide you with the .ino file I used for testing and and an .txt file containing the output on my machine. 
[Dckuino.js-4wDj.zip](https://github.com/Thecakeisgit/LocaleKeyboard.js/files/644677/Dckuino.js-4wDj.zip)

Nibot1, 2016-12-11:
With your Keyboard.h and *.cpp it works.
I don't know what I didi wrong, but OK.
Tank you for fast reply

Thecakeisgit, 2016-12-11:
Absolutely normal.. The **Keyboard.h** and **Keyboard.cpp** are the fix for the locale..
Fixed #14

closed 2016-12-11 by Thecakeisgit