Dukweeno/LocaleKeyboard.js #16: Help making a swedish keyboard layout

20repsej, 2017-01-18:
Hello!
I´m trying to make a locale for Swedish keyboard layout but I´m having problem with these keys: \, ^, ` and ~. I´m very noob at this so a little bit of help should be good.

gusgravious, 2017-05-04:
Hi there. Just started playing with a digispark and realized I have a problem with the keyboard layout as I also need it to work on Swedish keyboards. I am also a noob :-). I also too need to be able to compile and execute for keyboards on PC and Mac in swedish, meaning 2 slightly different layouts. I found my way here from Seytonics Malduino page (https://malduino.com/converter/info.html).

First of all, the Dckuino.js live version linked to in the instructions is dead (404) and I think it might be a good start if this works. I've found other live version available though I think the link should work anyway.

Second, have I got this right? One should compile using JS, run code through IDE to the device and execute. Characters not working correctly should be modified and cycle repeated. In other words, this is not a "permanent" modification in the IDE that will allow direct coding and compiling in the IDE, you will need to pass through the JS to get the correct key-mapping.

I'd love to contribute though I suppose my lack of knowledge and desperate lack of time might be a problem here. Hopefully there is a JPG attached to this message with 2 keyboards ((mac/pc) suppose some differences are possible moving from stationary to laptop though standard keys should be the same).

[image]

Let me know if and how I can help this progress!

// Gus

Nurrl, 2017-05-05:
Thanks @gusgravious, I'm trying to make an "auto-locale" maker, but do you know how to get the hex code a pressed key ?

gusgravious, 2017-05-05:
Hi @Nurrl, thanks for replying.

Not yet, I only looked for a coupIe of minutes this morning and found a project where a guy was trying to write a "translator" in C++ that accidentally printed the hex value instead of the key pressed. Somewhere else I found that one could print char variables as hex and dec in C# or something.

I'm not really a programmer and only have limited experience with python and BASH/AWK script though I'll get back to you if I manage to work something out.

Nurrl, 2017-05-05:
For me I'll go through Python, because we need to take full control of keyboard.. To get the keystrokes ^^

gusgravious, 2017-05-05:
I did think about Python as well, though I did find this after some quick "phone-googleing"

http://stackoverflow.com/questions/3381614/c-convert-string-to-hexadecimal-and-vice-versa

So this is obviously C++, I have not found a solution in python yet. What if you don't use keyboard input and just iterate through the "test-string", would the HEX output be system dependent (as the locale settings would likely be different?)? Hmmmm, well, I'll stop bugging you with this here and come back if I can muster up a suggestion instead.

Nurrl, 2017-05-05:
But it's not this hex value we need... The hex values for chars are static... The values we need are the hex keystrokes value :D @gusgravious (Aka. Keyboard raw output values)

gusgravious, 2017-05-05:
Ouch, that's embarrassing....  Well, it could have been a hole lot worse right, this could have been a public forum, oh wait.... 😰 I'll just put another coin in the ol' noob-jar, hehehe. I do feel kind of dumb right now though it is quite obvious of course. @Nurrl 

Should I miraculously manage to produce something worth sharing I'll get back to you, otherwise I'll stick to what I know (ctrl+c, ctrl+v) BTW, thank you for keeping your replies civilized

Nurrl, 2017-05-05:
You're welcome, there is only 3 not working chars, so you might just try to play with the altGrMap ;D
Obviously, this variable store what characters needs to be pushed with AltGr, and it works like this : 
  - `_altGrMap[23] = true;` means "*Use the AltGr modifier for the key index 23 = yes*"
And the key index is the position in the **asciiMap**, from 0 to 127...

Nurrl, 2017-05-05:
@gusgravious

gusgravious, 2017-05-05:
@Nurrl No thank you. Yep, this is likely something I can manage, and will solve my issue. If I get around to it ill try to figure out how to add this to the pile. Meanwhile, I did find this "[showkey](https://linux.die.net/man/1/showkey)", might be interesting and potentially something like this exists in python

Nurrl, 2017-05-05:
Yep, I knew showkey, but it doesn't output the good hex codes too, I don't know how to make this working .-.

Nurrl, 2017-07-04:
@loidor Here is the begin of what you want to do :D

closed 2017-07-05 by Nurrl