Dukweeno/LocaleKeyboard.js #19: The characters ^ and ` don't work in locale es_ES.

Thecakeisgit, 2017-01-18:
@Borja-Garduno Here we are, we'll fix this locale :D

Borja-Garduno, 2017-01-20:
I have been trying to achieve this, still without success. It really drives me nuts to have only two keys not working! Any ideas?
@Thecakeisgit 
http://stand.org/sites/default/files/styles/blog_post/public/Arizona/YourMyOnlyHope-vblog.png

Thecakeisgit, 2017-01-22:
Hmm, i'll need to modify the main code ^^ (Of Keyboard.cpp) and I'll notify you when ready ^^

Franco-Poveda, 2017-03-05:
> and < are not working for me, I have tried searching the code value with "showkeys" on linux and with "key codes" in mac.
[image]

@Nurrl can you give us a hint on what the error is in the code so we can try help you. Thx

Nurrl, 2017-03-06:
Actually, there is no bug in code, but according to @Borja-Garduno, you need to press **<** + **\<Space\>**, so we need to make a **_spaceMap** var..  
And I'm quite busy for now, so if you wan't to try this, feel free ;D

DiegoSepu95, 2017-04-11:
Did anyone fixed the issue I also have Franco-Poveda keyboard, and i cant type these symbols ~ < > \.
I edited the keyboard.cpp with no success.

Nurrl, 2017-04-11:
Nope, sorry man... @DiegoSepu95

BlueArduino20, 2017-04-11:
Any idea? I been trying to fix this problem without success...

DiegoSepu95, 2017-04-11:
@Franco-Poveda @BlueArduino20 
Yes i fixed it! here is the file.  [Keyboard.cpp](https://github.com/Nurrl/LocaleKeyboard.js/files/913833/Keyboard.cpp.txt) i fixed <>~.
Just change the file extension back to .cpp, that is for the spanish-ISO keyboard layout.

BlueArduino20, 2017-04-11:
@DiegoSepu95 I've tried your .cpp and it doesn't work at all. Look at this TheJedeX's repository: https://github.com/TheJedeX/Keyboard-Spanish.cpp/blob/master/Keyboard-Spanish.cpp
It has solved all the problems, but ^ and ` still not working.
After using the cpp I've put here:
Original text: !"#$%&'()*+,-./0123456789:;<=>?ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^`_abcdefghijklmnopqrstuvwxyz{|}~ @
Written text: !"#$%&'()*+,-./0123456789:;<=>?ABCDEFGHIJKLMNOPQRSTUVWXYZ[\][_abcdefghijklmnopqrstuvwxyz{|} @

BlueArduino20, 2017-04-11:
Ok, I've found the solution. We need to remove **_altGrMap[94] = true; // ^**, and **_altGrMap[96] = true; // `**. Here you've got my cpp code:

```
#include "Keyboard.h"

#if defined(_USING_HID)

//================================================================================
//================================================================================
//	Keyboard

static const uint8_t _hidReportDescriptor[] PROGMEM = {

  //  Keyboard
    0x05, 0x01,                    // USAGE_PAGE (Generic Desktop)  // 47
    0x09, 0x06,                    // USAGE (Keyboard)
    0xa1, 0x01,                    // COLLECTION (Application)
    0x85, 0x02,                    //   REPORT_ID (2)
    0x05, 0x07,                    //   USAGE_PAGE (Keyboard)
   
  0x19, 0xe0,                    //   USAGE_MINIMUM (Keyboard LeftControl)
    0x29, 0xe7,                    //   USAGE_MAXIMUM (Keyboard Right GUI)
    0x15, 0x00,                    //   LOGICAL_MINIMUM (0)
    0x25, 0x01,                    //   LOGICAL_MAXIMUM (1)
    0x75, 0x01,                    //   REPORT_SIZE (1)
    
  0x95, 0x08,                    //   REPORT_COUNT (8)
    0x81, 0x02,                    //   INPUT (Data,Var,Abs)
    0x95, 0x01,                    //   REPORT_COUNT (1)
    0x75, 0x08,                    //   REPORT_SIZE (8)
    0x81, 0x03,                    //   INPUT (Cnst,Var,Abs)
    
  0x95, 0x06,                    //   REPORT_COUNT (6)
    0x75, 0x08,                    //   REPORT_SIZE (8)
    0x15, 0x00,                    //   LOGICAL_MINIMUM (0)
    0x25, 0x65,                    //   LOGICAL_MAXIMUM (101)
    0x05, 0x07,                    //   USAGE_PAGE (Keyboard)
    
  0x19, 0x00,                    //   USAGE_MINIMUM (Reserved (no event indicated))
    0x29, 0x65,                    //   USAGE_MAXIMUM (Keyboard Application)
    0x81, 0x00,                    //   INPUT (Data,Ary,Abs)
    0xc0,                          // END_COLLECTION
};

Keyboard_::Keyboard_(void) 
{
	static HIDSubDescriptor node(_hidReportDescriptor, sizeof(_hidReportDescriptor));
	HID().AppendDescriptor(&node);
}

void Keyboard_::begin(void)
{
}

void Keyboard_::end(void)
{
}

void Keyboard_::sendReport(KeyReport* keys)
{
	HID().SendReport(2,keys,sizeof(KeyReport));
}

extern
const uint8_t _asciimap[128] PROGMEM;

#define SHIFT 0x80
const uint8_t _asciimap[128] =
{
  0x00,             // NUL          0
[... 240 lines cut ...]
}

void Keyboard_::releaseAll(void)
{
  _keyReport.keys[0] = 0;
  _keyReport.keys[1] = 0;
  _keyReport.keys[2] = 0;
  _keyReport.keys[3] = 0;
  _keyReport.keys[4] = 0;
  _keyReport.keys[5] = 0;
  _keyReport.modifiers = 0;
  sendReport(&_keyReport);
}

size_t Keyboard_::write(uint8_t c)
{
  uint8_t p = press(c);  // Keydown
  release(c);            // Keyup
  return p;              // just return the result of press() since release() almost always returns 1
}

Keyboard_ Keyboard;

#endif
```

Nurrl, 2017-04-11:
Okay ;D, it seems that this issue will be closed tomorrow ;D

Nurrl, 2017-04-11:
Thanks @DiegoSepu95 @BlueArduino20 :D

BlueArduino20, 2017-04-11:
@Nurrl I see a new problem... On spanish layout we need to press some key after ~ to write it. The problem comes when after **^** or **´** or **~** comes a space. To make ~ and after that, a space (~ ) we need to: ALTGR + 4, SPACE and SPACE.
Are there any way to add two keys? Something like **_0x21|SHIFT_** but not together at the same moment, but one after the another.

BlueArduino20, 2017-04-11:
**Here an example:**
Original text: ~ X
Written text: ~X

Nurrl, 2017-04-11:
I Think this is like all Latin-based keyboard, and it's different across OSs.. 
Example, for me in France : (**AltGr** + **2**) 
  - **Windows/OS X** -> **AltGr** + **2** + (Later)**Space** = ~
  - **Linux** -> **AltGr** + **2**  + (Later)**AltGr** + **2** = ~~

BlueArduino20, 2017-04-11:
Ok, then... There isn't any way to add keys (one after another), is there?

Nurrl, 2017-04-11:
Nope, sorry...

BlueArduino20, 2017-04-11:
Ok, thank you. Then I'll include a comment on my last code I've posted. How can I fix the es_ES file uploaded with the code I've posted?

Nurrl, 2017-04-12:
Oh sorry I did'nt saw, use the Wiki :D

BlueArduino20, 2017-04-12:
I've uploaded the file to the locales folder

Nurrl, 2017-04-12:
Yep, just modify the file as intended and send pull request :D

BlueArduino20, 2017-04-12:
Ok, It's done!

closed 2017-04-12 by Nurrl