Dukweeno/LocaleKeyboard.js #31: Need help creating a swiss keyboard layout

DerBuddie, 2017-04-16:
Hello!
I´m trying to make a locale for swiss keyboard layout but I'm not able to do it. Can someone here pls help me? :)

Nurrl, 2017-04-16:
Hi @DerBuddie,
Hmm do you have any picture of the keyboard ? ;D

DerBuddie, 2017-04-17:
http://www.lindy.ch/$WS/ld0101/websale8_shop-ld0101/produkte/medien/bilder/gross/21834_1.png

Nurrl, 2017-04-18:
Thanks @DerBuddie, and what keys don't work ? Please use this:
```vbs
STRING !"#$%&'()*+,-./0123456789:;<=>?ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^`_abcdefghijklmnopqrstuvwxyz{|}~ @
```
and compile with [Dckuino.js](https://nurrl.github.io/Dckuino.js/) __without selecting a locale(Copy/Paste)__, then you'll compare with this : 
```
!"#$%&'()*+,-./0123456789:;<=>?ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^`_abcdefghijklmnopqrstuvwxyz{|}~ @
```

They characters above are the only chars you can put with an Arduino, there is no exception, and we'll need to modify the Arduino Core to make this work, and this is not possible.

DerBuddie, 2017-04-18:
hi @Nurrl 
I get this:
+à*ç%/ä)=(`,‘.-0123456789éö;^:_ABCDEFGHIJKLMNOPQRSTUVWXZYü$¨&§?abcdefghijklmnopqrstuvwxzyè£!° „

Hope you can help me :)

max05051998, 2018-01-09:
Hello there @Nurrl @DerBuddie 
I would also be interested in a swiss keyboard layout and i wanted to ask if there has been any progress on the File. I would really like to help but i am not experienced in C++ but if there's anything i could do please tell me. :)
Thanks