Dukweeno/LocaleKeyboard.js #33: I will make one for the norwegian layout

Andsoeg, 2017-04-24:
I will make one for the norwegian layout, but i need to know how, lol. Any special apps or software i need to help me out?

Andsoeg, 2017-04-24:
do i need to make both cpp and .h file?

Nurrl, 2017-04-24:
Nop simply follow the guide in wiki: [How to create a locale ?!](https://github.com/Nurrl/LocaleKeyboard.js/wiki/How-to-create-a-locale-%3F!)

Nurrl, 2017-04-24:
Just .lang, just modify the sample.lang to fit with norwegian key layout ^^

Andsoeg, 2017-04-24:
STRING !"#$%&'()*+,-./0123456789:;<=>?ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^`_abcdefghijklmnopqrstuvwxyz{|}~ @

What should i do with this? I am sorry, not a very technical guy. Just someone who wants to have some fun on a norwegian keyboard

Andsoeg, 2017-04-24:
!"#$%&'()*+,-./012345STUVWXYZ[]^`_abcdefghijklmnopqrstuvwxyz{´}~@
[]^`_abcdefghijklmnopqrstuvwxyz{´}~@
This was the output

Andsoeg, 2017-04-24:
This was based of the danish one. or should i build from scratch?

Andsoeg, 2017-04-24:
Yeah ok, i have found out everything except the part of how i could know the "Coordinate" to the number/character/symbol.

  0x1e|SHIFT,     // !
  0x1f|SHIFT,     // " ,
  0x31,          // #

This is how it is regularly.  This symbol (") is to the right for (!) on my keyboard.
Hashtag is to the right again for (!), then i assumed that # should be 0*1g I SHIFT
But no, (
exit status 1
unable to find numeric literal operator 'operator""g'
)
This is the error. This whole coordinate thing confuses me

Andsoeg, 2017-04-24:
Step 3 - Time to test
is also unclear to me

Andsoeg, 2017-04-25:
!"#$%&'()*+,-./0123456789:;<=>?ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^`_abcdefghijklmnopqrstuvwxyz{|}~@
I got this output now

Andsoeg, 2017-04-25:
All i miss is the space between ~@

Nurrl, 2017-04-25:
Hi, you made several mistakes in you posts, let me explain:
  - `0x1g` and `0*1g` are completely different, the first mean `1g` in hex, the second mean `0[multiplication]1g` .
 - `0x1g` is not a valid hexadecimal number, the characters range are 0->9 - a->f, for a total of 16 chars.

__Step 3__ is when you want to test, obviously, you use Dckuino.js, then you upload the code in the Arduino, you open a text editor, and you plug the Arduino to see what it will write ;D

Last thing, what keys aren't working now ? Please list them out ^^

Nurrl, 2017-04-25:
To convert characters to hex, for now i didn't found anything... @Andsoeg

Andsoeg, 2017-04-25:
Only character not working is spacebar.

Nurrl, 2017-04-28:
Oh, so find the line for space ^^ And try :D @Andsoeg

Nurrl, 2017-04-28:
*To fix it ^^

Andsoeg, 2017-04-29:
I have, but it ios like 100 diffrent hex chars to test, and i doesnt seem to find it

Nurrl, 2017-04-30:
Are you sure that `STRING This is a test`, outputs: `Thisisatest` ?

Andsoeg, 2017-05-08:
Yes, what technique do you use when you try to find characters?

Andsoeg, 2017-05-08:
/*
  Keyboard.cpp

  Copyright (c) 2015, Arduino LLC
  Original code (pre-library): Copyright (c) 2011, Peter Barrett

  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General Public
  License along with this library; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
*/

#include "Keyboard.h"

#if defined(_USING_HID)

//================================================================================
//================================================================================
//  Keyboard

static const uint8_t _hidReportDescriptor[] PROGMEM = {

  //  Keyboard
    0x05, 0x01,                    // USAGE_PAGE (Generic Desktop)  // 47
    0x09, 0x06,                    // USAGE (Keyboard)
    0xa1, 0x01,                    // COLLECTION (Application)
    0x85, 0x02,                    //   REPORT_ID (2)
    0x05, 0x07,                    //   USAGE_PAGE (Keyboard)

  0x19, 0xe0,                    //   USAGE_MINIMUM (Keyboard LeftControl)
    0x29, 0xe7,                    //   USAGE_MAXIMUM (Keyboard Right GUI)
    0x15, 0x00,                    //   LOGICAL_MINIMUM (0)
    0x25, 0x01,                    //   LOGICAL_MAXIMUM (1)
    0x75, 0x01,                    //   REPORT_SIZE (1)

  0x95, 0x08,                    //   REPORT_COUNT (8)
    0x81, 0x02,                    //   INPUT (Data,Var,Abs)
    0x95, 0x01,                    //   REPORT_COUNT (1)
    0x75, 0x08,                    //   REPORT_SIZE (8)
    0x81, 0x03,                    //   INPUT (Cnst,Var,Abs)

  0x95, 0x06,                    //   REPORT_COUNT (6)
    0x75, 0x08,                    //   REPORT_SIZE (8)
    0x15, 0x00,                    //   LOGICAL_MINIMUM (0)
    0x25, 0x65,                    //   LOGICAL_MAXIMUM (101)
    0x05, 0x07,                    //   USAGE_PAGE (Keyboard)

  0x19, 0x00,                    //   USAGE_MINIMUM (Reserved (no event indicated))
    0x29, 0x65,                    //   USAGE_MAXIMUM (Keyboard Application)
    0x81, 0x00,                    //   INPUT (Data,Ary,Abs)
    0xc0,                          // END_COLLECTION
};

Keyboard_::Keyboard_(void)
{
  static HIDSubDescriptor node(_hidReportDescriptor, sizeof(_hidReportDescriptor));
  HID().AppendDescriptor(&node);
}

void Keyboard_::begin(void)
{
[... 259 lines cut ...]
  return 1;
}

void Keyboard_::releaseAll(void)
{
  _keyReport.keys[0] = 0;
  _keyReport.keys[1] = 0;
  _keyReport.keys[2] = 0;
  _keyReport.keys[3] = 0;
  _keyReport.keys[4] = 0;
  _keyReport.keys[5] = 0;
  _keyReport.modifiers = 0;
  sendReport(&_keyReport);
}

size_t Keyboard_::write(uint8_t c)
{
  uint8_t p = press(c);  // Keydown
  release(c);            // Keyup
  return p;              // just return the result of press() since release() almost always returns 1
}

Keyboard_ Keyboard;

#endif

Andsoeg, 2017-05-08:
I dont know any about this shit, and my brain went full overload during this time. But however it seems like it is the test string something is wrong with. because i can type regular words with a space between, this should be perfect?

Andsoeg, 2017-05-08:
This is the Norwegian keyboard.cpp as i mentioned