Dukweeno/LocaleKeyboard.js #34: I've created an automatic .cpp generator

BlueArduino20, 2017-05-07:
The software I've made generates a Keyboard.cpp according to our locale chosen.
This is the main idea: https://www.youtube.com/watch?v=gxE7GKLcC2w
- You can use this software in Linux: https://github.com/BlueArduino20/LocaleKeyboard.SH
- Or Windows: https://github.com/BlueArduino20/LocaleKeyboard.BAT

Nurrl, 2017-05-11:
Hi, @BlueArduino20, thanks for doing this, but why ? The LocaleKeyboard.js does the same thing, why not creating an index.html ? I don't really understand why you did this ;D
Thanks anyway...

BlueArduino20, 2017-05-11:
I think that it's easier using the command line (And more friendly for pentesters)

Nurrl, 2017-05-11:
Pentesters could use npm :D

BlueArduino20, 2017-05-18:
Right, but if we've got more ways to do something, better. It's about variety :). Anyway, thank you for doing this repository!

Nurrl, 2017-05-18:
No problem ^^
Maybe improving your project could be great, because I saw you hard-coded some things like locale selection :S

BlueArduino20, 2017-05-19:
Take a look to the new locale selection: https://github.com/BlueArduino20/LocaleKeyboard.BAT/blob/master/LocaleKeyboard.BAT

exanor, 2018-04-08:
n