Dukweeno/LocaleKeyboard.js #35: ES_LAT?

jalax0990, 2017-05-09:
How i can modify ES_es keyboard to es_lat??

Nurrl, 2017-05-11:
Hi, what keys aren't working ? The project only supports the non-extended ascii table : 
[image]

jalax0990, 2017-05-12:
If i print this string in Es_ES  i print other thing

!"#$%&'()*+,-./0123456789:;<=>?ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^`_abcdefghijklmnopqrstuvwxyz{|}~ @

This is i get

!"$%&'()*+,-./0123456789:;<=>?ABCDEFGHIJKLMNOPQRSTUVWXYZ¬~¨´_abcdefghijklmnopqrstuvwxyz^`

Nurrl, 2017-05-12:
So, apparently, you miss a "AltGr" keys, how do you type these keys on you keyboard ? : # [ ] \ ` { | } ~ @ ?

jalax0990, 2017-05-16:
This is the Keyboard por example

 SHIFT+3 = # 
ALTGR+q = @
ALTGR+' = \

https://k60.kn3.net/E/6/6/C/F/6/21B.png

Nurrl, 2017-05-16:
Can you send a pull request, so I can help you ? By modifying directly the code ?

*Note: Use this guide -> https://github.com/Nurrl/LocaleKeyboard.js/wiki/How-to-create-a-locale-%3F!*

jalax0990, 2017-05-18:
If i try to make a keymap i get this

> 
keymap = ^
 '32':' ',
 '33':'!',
 '34':'"',
 '35':'',
 '36':'$',
 '37':'%',
 '38':'&',
 '39':''',
 '40':'(',
 '41':')',
 '42':'*',
 '43':'+',
 '44':',',
 '45':'-',
 '46':'.',
 '47':'/',
 '48':'0',
 '49':'1',
 '50':'2',
 '51':'3',
 '52':'4',
 '53':'5',
 '54':'6',
 '55':'7',
 '56':'8',
 '57':'9',
 '58':':',
 '59':';',
 '60':'|',
 '61':'=',
 '62':'°',
 '63':'?',
 '64':'',
 '65':'A',
 '66':'B',
 '67':'C',
 '68':'D',
 '69':'E',
 '70':'F',
 '71':'G',
 '72':'H',
 '73':'I',
 '74':'J',
 '75':'K',
 '76':'L',
 '77':'M',
 '78':'N',
 '79':'O',
 '80':'P',
 '81':'Q',
 '82':'R',
 '83':'S',
 '84':'T',
 '85':'U',
 '86':'V',
 '87':'W',
 '88':'X',
 '89':'Y',
 '90':'Z',
 '91':'',
 '92':'^',
 '93':'~',
 '94':'',
 '95':'_',
 '96':'',
 '97':'a',
 '98':'b',
 '99':'c',
 '100':'d',
 '101':'e',
 '102':'f',
 '103':'g',
 '104':'h',
 '105':'i',
 '106':'j',
 '107':'k',
 '108':'l',
 '109':'m',
 '110':'n',
 '111':'o',
 '112':'p',
 '113':'q',
 '114':'r',
 '115':'s',
 '116':'t',
 '117':'u',
 '118':'v',
 '119':'w',
 '120':'x',
 '121':'y',
 '122':'z',
 '123':'^',
 '124':'',
 '125':'`',
 '126':'',
 '127':'',
 '128':'',
 '129':'',
 '130':',
 '131':'',
 '132':'',
 '133':'',
 '134':'',
 '135':'
 '136':'',
 '137':'',
 '138':'',
 '138':'',
 '139':'',
 '140':'a',
 '141':'b',
 '142':'c',
 '143':'d',
 '144':'e',
 '145':'f',
 '146':'g',
 '147':'h',
 '148':'i',
 '149':'j',
 '150':'k',
 '151':'l',
 '152':'m',
 '153':'n',
 '154':'o',
 '155':'p',
 '156':'q',
 '157':'r',
 '158':'s',
 '159':'t',
 '160':'u',
 '161':'v',
 '162':'w',
 '163':'x',
 '164':'y',
 '165':'z',
 '166':'1',
 '167':'2',
 '168':'3',
 '169':'4',
 '170':'5',
 '171':'6',
 '172':'7',
 '173':'8',
 '174':'9',
 '175':'0',
 '176':'',
 '177':'',
 '178':',
 '179':'	',
 '180':' ',
 '181':''',
 '182':'¿',
 '183':'´',
 '184':'+',
 '185':'}',
 '186':'}',
 '187':'ñ',
 '188':'{',
 '189':'|',
 '190':',',
> 

But i can get the other characters i im lost.

 I try to follow the tutorial but i can get the lost characters 
!"$%&'()*+,-./0123456789:;<=>?ABCDEFGHIJKLMNOPQRSTUVWXYZ¬~¨´_abcdefghijklmnopqrstuvwxyz^`

Nurrl, 2017-05-18:
This really not what you need to do x) You need to get something like > [This](https://github.com/Nurrl/LocaleKeyboard.js/blob/master/locales/es_ES.lang) <