Dukweeno/LocaleKeyboard.js #36: German Keyboard Bug

ConTRoniiX, 2017-05-31:
Hello,
there's a bug the german keyboard. If I want to let it type a backslash(\) it types a & instead.
pls fix xD

ConTRoniiX

Nurrl, 2017-05-31:
Hi,
How do you type this ?

ConTRoniiX, 2017-06-01:
If I want to type a backslash, I would do this:
`   
    delay(defaultDelay);
    Keyboard.print("  \\");
    
    /* ----- End-Script -----*/
    
    Keyboard.end();
  }
}

void loop(){
  ledOn = !ledOn;
  digitalWrite(ledPin, ledOn);
  delay(blinkInterval);
}`

ConTRoniiX, 2017-06-01:
Kinda messed up the code thingy.

Nurrl, 2017-06-01:
When I said how, I meant on your keyboard :3

ConTRoniiX, 2017-06-02:
oh :D i press ALT GR and then the ß

ConTRoniiX, 2017-08-02:
Hello? Is it fixable?

Nurrl, 2017-08-02:
Maybe, but as I did for French, I tested manually.. And it was painful .-. @ConTRoniiX

ConTRoniiX, 2017-08-02:
Well I am sorry for that^^ I opened a new Issue with other Bugs I found, let me know how I can help you

closed 2017-08-02 by Nurrl