Dukweeno/LocaleKeyboard.js #37: Danish: Alt Gr

Jackiedk100, 2017-06-01:
Hello,
All Alt Gr symbols doesn't work.

Example: Alt Gr + 4 = $
but in this code:
Alt Gr + 4 = 4

Nurrl, 2017-06-01:
Hi, thanks for your submit ^^,
Please can you send me the code,
provide me what way you used to get the script(**Copy/paste** or **Download button**)
and if you downloaded, what locale you've used ^^

Jackiedk100, 2017-06-01:
Everything is from:
https://malduino.com/converter/
da_DK locale

Jackiedk100, 2017-06-01:
Thanks for the fast respond

Nurrl, 2017-06-01:
No problem, but I saw that they didn't used my LocaleKeyboard.js, I don't know what they did, they just took all the .lang.. Maybe you'll have more by sending the issue to them ^^

Jackiedk100, 2017-06-01:
Do you think that is why?

Nurrl, 2017-06-01:
I don't even know if they got the same .lang as me..