Dukweeno/LocaleKeyboard.js #38: Problem with '<>; Ç ^ ~}' in Layout pt_BR.

thiagofw, 2017-06-04:
Hello, I created the keyboard.cpp file according to the documentation, I wrote the code and after uploading to arduino (arduino micro - leonardo) I am not having success using punctuation, accentuation as required by the Portuguese language of Brazil.
I wonder if anyone else has or has had this problem or if someone could give me an aid in correcting this.
Ex:. :; ^~ `'{ [ } ]
Does not work correctly, incidentally, does not work, leaving spaces blank or simply empty.

Nurrl, 2017-06-16:
Hi, sorry for the late answer, can you provide your .lang and a picture of your keyboard ? @thiagofw

diegochaves, 2017-08-30:
Just started one pt_BR.lang, so far i just came here to see if someone is doing it.

https://en.wikipedia.org/wiki/Portuguese_keyboard_layout#/media/File:KB_Portuguese_Brazil.svg