Dukweeno/LocaleKeyboard.js #43: Issues in listLocale

loidor, 2017-07-05:
While adding Swedish, I noticed that listLocale is... a bit weird. It's not really in any order, and they use locale codes which messes a bit with things since keyboard layouts don't follow locale codes exactly.

The languages right now are:

```
United States       us
United Kingdom      uk
Germany             gr
French              fr
Belgium             be *
Spain               es
Finland             su
Czech Republic      cz
Denmark             dk
Portugal            po
Turkey              tr
Italy               it
Sweden              sv
```

\* I'm not sure about Belgian Keyboards. In the PR this is specified as Belgian dot (I find Belgian Period when googling about, so I guess that's the same), and then there's Belgian Comma, but according to the IBM paper I got all this from (http://www.borgendale.com/keyboard.pdf) there's just one Belgian layout. Very confusing. Perhaps @KevinDenys can shed some light on this?

Also, sample is listed as a language - I'm not sure if this is a debugging thing and supposed to be like that or not. 

My suggestion is to rename them all to the correct keyboard layout names, remove sample from listLocale and adding a part in the wiki about what keyboard layout codes there are or something.

Another thing would be to do some kind of alphabetically sorting (but still keeping English on top), but then I guess they need to be presented better too - i.e. Finland rather than su.

If it's possible to use spaces in the .lang file names maybe the country names would be even better. United States.lang and United States in listLocale.

What do you think @Nurrl?

KevinDenys, 2017-08-06:
Didn't see you mentioned me, euhm I can remember making the Belgian file was a pain in the *ss, lots of trial and error. Everything I found online didn't match. The easiest part was switching qwerty -> azerty. The other keys like a dot, plus, minus were looking at the Qwerty version and adding a shift in front of it. (If I can remember correctly)

Nurrl, 2017-11-09:
Hi @loidor, this project is about to be merged directly into Duckuino, and I'll be using these locales codes, thanks for your help and sorry for the late answer ^^

loidor, 2017-11-10:
Great!

closed 2017-11-10 by loidor