Dukweeno/LocaleKeyboard.js #44: Repost of "German Keyboard Bug + other Bugs

ConTRoniiX, 2017-08-02:
Hello,

I didn't get an answer on my last Issue, so I thought i should repost it.

Bugs I found for the german keyboard:

\ is instead ß. I write it like this: ALT GR + ß
@ is instead q. I write it like this: ALT GR + Q
[ is instead 8 and ] is instead 9. I write them like this: SHIFT + 8 and SHIFT + 9
| is instead <. I write it like this: ALT GR + <

I hope you can fix it. I found a way to get around the first error (not using the backslash), but the other ones are unreplacable.

ConTRoniiX

Nurrl, 2017-08-02:
I answered 38min ago on the other, I'll close it.

Nurrl, 2017-08-02:
You could use logic, To type [, you type SHIFT + 8, so you take the value of 8(0x25) and you apply SHIFT so : **[** -> **0x25|SHIFT**

Nurrl, 2017-08-02:
Using this logic, you can guess the other ;)

ConTRoniiX, 2017-08-02:
ok, so ALT GR is just for example: "value of <|ALT GR"

Nurrl, 2017-08-02:
No, I couldn't do this because of Arduino Core limitations.. So i made the _altGrMap : 
You count from 0(the first item in list is id 0) to what you want and you can apply an ALTGR by setting _altGrMap[<theidyoufindbycounting>] = true;

ConTRoniiX, 2017-08-02:
ok thanks, i'll try it

ConTRoniiX, 2017-08-04:
The characters I mentioned are already included:

```
  _altGrMap[126] = true; // ~
  _altGrMap[123] = true; // {
  _altGrMap[91] = true;  // [
  _altGrMap[93] = true;  // ]
  _altGrMap[125] = true; // }
  _altGrMap[92] = true;  // bslash
  _altGrMap[124] = true; // |
  _altGrMap[64] = true;  // |
```

And I wrote them for example like this;

`Keyboard.print("\");`

Did I do something wrong? 
It still doesn't print those Characters

Nurrl, 2017-08-07:
Hi @ConTRoniiX, did you tried with all the chars ?

ConTRoniiX, 2017-08-07:
Yes, I did. And none of the ones I listed at the beginning is working.

ConTRoniiX, 2017-08-29:
Would be nice if you would help me, I'm still having the same problem.

No hate, I really love your project and I think it is awesome that you helped me so much at my past messages and you probably just missed my last one ;)

Nurrl, 2017-08-29:
Hi, yep I missed it:
'\\' may not be fixable, because we apply the altGr on the key you use with altGr, for example for '\\' you may fix it like this : 
```c
const uint8_t _asciimap[128] =
{
/* ... */
  <Hex code for 'ß'>, // bslash
/* ... */
};

void initAltGr() {
  _altFine = true;
  /* ... */
  _altGrMap[92] = true; // bslash
  /* ... */
}
```
Because you do 'ß' + altGr. This apply for all 'altGr keystrokes'.

For other, I need your complete .lang :D