Dukweeno/LocaleKeyboard.js #49: Strings on danish keyboard

Jackiedk100, 2018-03-07:
There are many errors with symbols in the danish locale file.

This is all the numbers and symbols at the numbers - "1234567890+´"
This is shift + numbers - "!"#¤%&/()=?"
This is alt + gr + 23457890´ - "@£$€{[]}|"
This is the symbol left for 1 - "½"
This is shift + the symbol left for 1 - "§"

This is all the letters from qaz and to the right. (you also don't have æøå) 
"qwertyuiopå" + "¨" <-- this have to get pressed twice or with a following space.
"asdfghjklæø'"
"<zxcvbnm,.-"

With shift on the key left for z - ">"
With alt + gr on the key left for z - "\"
With shift on the key right for m - ";"
With shift on the key two right for m - ":"
With shift on the key three right for m - "_"
With shift on the key right for ø - "*"
With shift on the key right for å - "^" (have to be pressed twice or with a following space)
With alt + gr on the key right for å - "~" (have to be pressed twice or with a following space)

Jackiedk100, 2018-03-07:
It's this one 
https://github.com/Nurrl/LocaleKeyboard.js/blob/master/locales/da_DK.lang
or
https://github.com/Seytonic/malduino/blob/master/converter/src/locales/da_DK.lang

Jackiedk100, 2018-03-07:
This is how it appears right now:

This is all the numbers and symbols at the numbers - "1234567890+ "
This is shift + numbers - "!"#y%&/()=?"
This is alt + gr + 23457890  - "2x47890´"
This is the symbol left for 1 - "½"
This is shift + the symbol left for 1 - "2"

This is all the letters from qaz and to the right. (you also don't have 65)
"qwertyuiop5" + "3" <-- this have to get pressed twice or with a following space.
"asdfghjkl6'"
"<zxcvbnm,.-"

With shift on the key left for z - ">"
With alt + gr on the key left for z - ""
With shift on the key right for m - ";"
With shift on the key two right for m - ":"
With shift on the key three right for m - "_"
With shift on the key right for  - "*"
With shift on the key right for 5 - "^" (have to be pressed twice or with a following space)
With alt + gr on the key right for 5 - "¨" (have to be pressed twice or with a following space)

Nurrl, 2018-03-07:
Hi @Jackiedk100, please note that this project is for Duckuino, not Malduino, which is a fork and we **don't** maintain.
Please identify problems only on *https://github.com/Nurrl/LocaleKeyboard.js/blob/master/locales/da_DK.lang*.
Thanks !

Jackiedk100, 2018-03-07:
How do I do that?

Nurrl, 2018-03-07:
With what project did you compiled your code ? (On what site ?)

Jackiedk100, 2018-03-07:
The file from LocaleKeyboard.js for Denmark

Nurrl, 2018-03-07:
LocaleKeyboard.js can't print characters alone, so with what website you've tested this ?