Dukweeno/LocaleKeyboard.js #50: French Canadian Keyboard

Build-0-Matic, 2018-08-18:
Hi!
I'm trying to make a locale for a French-Canadian keyboard and I'm having a bit of trouble figuring things out.
I've spent most of the afternoon modifying code-updating-trying-redo from start....
We have a lot important characters that use the AltGr key. Like \@~[]{}´­ and others that need to type a key followed by a space like ^

I noticed that the _altGrMap is needed but can't figure out that the number between [ ] is.

Nurrl, 2018-08-19:
I'll close the Duckuino issue, we'll continue here :D

Build-0-Matic, 2018-08-19:
Like you asked, I did read the wiki.
So far, I've managed to figure out the following (I think).
The asciimap locates each character on the physical keys using the BIOS key calls in hex.
The asciimap's order is important because it sequentially maps the keys.
For the keys that are not at their right place need to get their hex address changed.
Those that need the AltGr key need to have their position in the list mentioned starting the count at 0 with   0x00,             // NUL

Am I getting this right?
yesterday when I posted I had been trying all afternoon to figure this out and had a massive headache...

Nurrl, 2018-08-19:
Yes, you're perfectly right, the altGrMap is a trick I invented to fool the original Keyboard library. It's a bit messy and tricky to do but it does work.
You could try to use the French locale to adapt the French-Canadian.

Nurrl, 2018-08-19:
I agree that's tricky and long, but I haven't figured out another way to do it :c I you have any idea !

Build-0-Matic, 2018-08-19:
The way that would be the easiest would be to do like the shift key trick but using the right alt key or ctrl+left alt....

Yesterday I stumbled onto this https://handmade.network/forums/t/2011-keyboard_inputs_-_scancodes,_raw_input,_text_input,_key_names
which might help.

Nurrl, 2018-08-19:
It will be okay for the memory, but I can't tell if it will just don't work because of how the Arduino Core uses it !

Nurrl, 2018-08-19:
Eating 4bytes is not really a big deal :D

Build-0-Matic, 2018-08-19:
Right now I'm sorting through all the keys and commenting those that need the AltGr key...
I'm also running into a weird issue with the first  0x2c,          //  ' ' entry... it seems to eat-up other characters...
But back to the the big mod... what would it take to increase the asciimap size?

Nurrl, 2018-08-19:
I don't know, I need to investigate more ^^ For 0x2c, I don't know, but you're maybe printing a Backspace !

Build-0-Matic, 2018-08-19:
But since it's at the start and doesn't affect the english keyboard, can't possibly be that...
Looks like it's modifying forward...

Nurrl, 2018-08-19:
Hmm, so I don't know .-.

Build-0-Matic, 2018-08-19:
What's the first character of the sequence supposed to be?
Maybe it just jets mapped in a bad way?

Build-0-Matic, 2018-08-19:
Just found this...
https://gist.github.com/MightyPork/6da26e382a7ad91b5496ee55fdc73db2

Build-0-Matic, 2018-08-19:
And the first character before the ! is a space I presume?
So far, I'm getting this thing down. Here's the test sequence I get
"#$%&'()*+,-./0123456789:;<=>?ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^`_abcdefghijklmnopqrstuvwxyz{|}~"

Build-0-Matic, 2018-08-20:
Finally fixed the first missing characters problem. It turns out the Arduino was to fast to start typing. I increased the delay to 1000ms and that fixed it.
But now I'm having a few problems with the backslash... it works intermittent.....
my test string now looks like this
!"#$%&'()*+,-./0123456789:;<=>?ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^`_abcdefghijklmnopqrstuvwxyz{|}~ @
Probably is good enough to use as is... but still it bugs me a bit.

Nurrl, 2018-08-20:
Nice, there's only a problem with `\`

Build-0-Matic, 2018-08-20:
Yes, I know but for some reason can't figure this one out.
the first \ doesn't print, but the second or third does... this really puzzles me.

Nurrl, 2018-08-20:
So it works, it's just the delay I guess :D

Build-0-Matic, 2018-08-20:
Well, at first I was not getting the `!` to print. That was delay related.
But I still can't get the `\` to print reliably... weird.
But for all the rest, it works pretty OK, so I made a lang file and submitted it anyways.

Nurrl, 2018-08-20:
Nice :D Thanks for your work for the project ^^

Build-0-Matic, 2018-08-20:
You're welcome🙂

Nurrl, 2018-08-20:
For future reference, we'll use the Duckuino project since `LocaleKeyboard.js` will be '''bundled''' into Duckuino. No worries, I'll adapt your locale for Duckuino.

Nurrl, 2018-08-20:
Feel free to send your pull request :D

Build-0-Matic, 2018-08-20:
Will do.